darknao/btClubSportWheel #1: Teensy-lc Custom Firmware for Fanatec Wheel conflicting with CSPv3 driver

frapstered, 2016-09-05:
Hello.
First of all - big thanks for this project. I did select this as my path for converting all my fanatec gear for OSW and it's working really well, except I'm experiencing an issue, where when I open Fanatec Wheel game controllers applet proprty page, with now (after real fanatec base has been removed) basic Windows controller driver user interface with buttons and axis, which all work fine, but if i leave and enter Fanatec ClubSport Pedals pages, I now would get also the basic window driver pages, not the pretty and with additional configuration options pages/firmware updates fialogs installed with fanatec wheel driver for the pedals.
This also creates an issue with the pedals in some games, thus it forces me to always make sure that fanatec driver is what I get any time I load Fanatec Pedals property pages.
This can be worked around by uninstalling and reinstalling fanatec driver and **not** entering Fanatec Wheel (teensy connected hub) basic property pages. For once you enter Fanatec Wheel properties (which will often be the full fanatec driver broken interface) and then access the pedals, they will get the basic driver UI again and would need fanatec driver reinstallation to rake place.
Is there any way that yourself, or someone who knows the process could update your custom firmware, so that it would not be conflicting with installed fanatec wheel driver used by the pedals? I assume it's an issue either because of the Fanatec Wheel naming or perhaps even a matching GUID of the fanatec real wheel in the firmware -I'm only guessing.
I have very limited knowledge or coding skills, but if there was a template/and an obvious enough way to obtain and make edits to your otherwise great teensy-lc usb firmware and compile it, I'd be up for it myself.
Apologies about the long report and slightly odd sentence structure.

darknao, 2016-09-07:
Thank you for your bug report.
Unfortunately, I don't own any Fanatec gear right now to see what exactly is going on here. So I can't promise this will be fixed any time soon.

But you assume right, as I had to fool Fanaleds by using the same VID/PID as the real thing to get it working.
Using another PID should fix your issue, but by doing that, you'll then loose fanaleds support (unless you force it to use the new PID).

I assume the Fanatec driver recognize the wheel as a real one, and try to communicate in a way the wheel doesn't understand.

frapstered, 2016-09-13:
Many thanks for you reply.

No problem, understood. I don't need fanaleds, personally,  so would be very happy to change the PID to something else.
How easy would it be for me to do that?
I of course checked your hex file for any obvious strings in a hex/binary editor, but I fear it may not be as easily hackable, as I'm not recognising anything obvious, so a pointer or an advice would be appreciated!

frapstered, 2016-09-20:
Hello again

I have some additional problem facts. Some games really do get confused about multiple steeringcwheels detected and some switch to  "fanatec wheel" by default, creating more conflict, to the point where the only resolution is to always unplug converted hubs and rims, losing paddle shiftin/buttons.

Is there anyone at all who would be able to change the PID and compile new firmware hex for those of us with this issue? 
I'd be very happy to, but my knowledge isn't strong enough what and where to change
Thanks in advance!

darknao, 2016-09-20:
Hi,

You can easily modify the PID here:
https://github.com/darknao/btClubSportWheel/blob/master/teensy3/usb_desc.h#L139
You'll have to use an unused PID by Fanatec to minimize conflict, something like 0xbeef should do the job.

If you don't know how to compile a new firmware and don't mind to wait a little, I may have some time to do it for you later in the day.

frapstered, 2016-09-20:
Many thanks for that! I will happily wait for your file.

But thanks for the pointer, I may use the info if all else fails :)

best regards

darknao, 2016-09-20:
here you go: [firmware_for_teensyLC](https://phunk.me/LZO1CC?dl=1)
It's an untested firmware, so keep in mind it may not work as expected.
Note: It's for Teensy LC only (not 3.x)

frapstered, 2016-09-21:
Hello again

I was delaying trying the update, due to the need to undo all 4 of my devices using this teensy method, but today the problem with the pedals God even worse, where games and Windows would not initial use the pedals at all.
So this presented the perfect opportunity to test new firmware, and I'm happy to report it seems to have resolved all my issues.
Unless something else crops up, I thank you kindly and will consider the issue resolved
Best regards

frapstered, 2016-09-22:
All is good, so closing the issue
Thanks again

closed 2016-09-22 by frapstered