darknao/btClubSportWheel #4: Fanatec to Fanatec

NMRO, 2018-05-21:
Hello
I have a Fanatec Base, and I would like to know if your project works if I connect it directly to a Fanatec base?
I would like to use other hoops, but maybe it does not work
regards

darknao, 2018-05-22:
Hi NMRO. This conversion allow you to connect Fanatec Wheels directly to your computer using USB, and thus, to any wheel base (assuming you have the matching quick release adapter).
You cannot use it to connect non-Fanatec wheel to a Fanatec Base.
What you need for that is the Fanatec Universal Hub.

0i8u2, 2018-05-29:
Hi, sorry if this is in the wrong place.
Can this conversion be used on the new Fanatec mclaren gt3 wheel ?
Also thanks for updating the USB files, has fixed my double shift problem that was get worse for some reason. Wasn’t till I started playing rfactor 2 and rre I first noticed it.

rich1015, 2018-05-29:
I think you can, but you need to compile your own hex file using the files of the testing branch - which I'm currently still failing at, but time will tell me the way to do so

closed 2018-05-29 by NMRO

lshachar, 2020-02-10:
@NMRO, [here's what you're looking for](https://github.com/lshachar/Arduino_Fanatec_Wheel)