darknao/btClubSportWheel #5: Wrong button presses registering

Hugorruss, 2018-05-25:
I appear to be having a problem with the wrong button presses registering.  I am using a Fanatec Clubsport BMW GT wheel in Windows 10.

For example If I shift up (button 16) with the shifter it sometimes registers the purple button (4) being hit .

The problem also appears when I use the little joystick on the right of my wheel.  If I hold the joystick in one position and look at the Windows diagnostic tool for the joystick, I can see it also is triggering a bunch of other buttons.  Several different buttons flash on and off while holding the joystick in position.

darknao, 2018-05-25:
Hey Hugorruss

Most of the times, this behavior is caused by either some EMI issue, or loose connection on the Teensy board.
If you still have this issue when the wheel isn't connected to the base, then it's a more likely a bad connection.
Try fiddling the cables between the Teensy and the wheel pcb while looking at your diag tool on windows (i recommend using DIView or JoystickTest) and check for any random button presses to confirm this diagnostic.

Did this issue came from nowhere one day, or never worked correctly since the beginning?

Hugorruss, 2018-05-28:
Hey darknao,

Thanks!  It looks like it's an EMI issue.  If I turn off my OSW, it works perfect.  I hold the joystick to one side and only that input registers.  If I turn on my OSW and hold the joystick to one side, the other buttons start flashing like christmas lights.  

I wrapped the teensy in electrical tape before putting it all together, do you have any other tips for making this go away?

Hugorruss, 2018-05-28:
Hey Darknao,
OK, I ran a wire from one of the screws on my PC that holds in the power supply to the mount for my MIGE motor and all EMI issues are gone now.  Maybe not the most elegant solution but super happy it was an easy fix, thanks for pointing me in the right direction!

closed 2018-05-28 by Hugorruss

darknao, 2018-05-28:
Glad you get this sorted out!
Drive safe :)