darknao/btClubSportWheel #6: Problem McLaren Clutch

PFC-Paula1969, 2018-06-11:
**Describe the bug**
A clear and concise description of what the bug is.

**Configuration (please complete the following information):**
 - Wheelbase: [e.g. OSW]
 - Rim: [e.g. BMW GT2, Formula]
 - Teensy Board: [e.g. LC, 31 or 32]
 - Firmware type: [e.g. BT, USB]
 - Firmware version, if known: 08062018

**Additional context**
Add any other context about the problem here.
After a while, the clutch is gone. In the game, in the win game controller and in the diview. Disconnect and reconnect, then it's time again. Until it fails

rich1015, 2018-06-11:
I can confirm the issue. 
After approx. 30 minutes I lose all inputs of the clutch paddles, no matter what mode (A to D) is set. 

Wheelbase not connected, McLaren Wheel, Teensy LC, USB, v1.2-beta.1

rich1015, 2018-06-13:
A small heads-up:

rich1015, 2018-06-13:
A small heads-up:
It seems that an outdated Fanatec firmware is causing this as both PFC-Paula1969 and I (and others we talked with) use their McLaren wheel "as delivered" and did never update the original firmware -> thus we're still at v1.9(?)

darknao, 2018-06-16:
This is a known bug, and I agree that it seems related to the original Fanatec firmware (v19) shipped with the wheel.
Fanatec has released a firmware update day one, and people using it doesn't seems to be affected by this issue.
Unfortunately, there is no known way of updating the wheel without a Fanatec wheel base.