darknao/btClubSportWheel #7: Formula rim has an issue with new beta firmware.

jamesdio, 2018-07-02:
**Describe the bug**
A clear and concise description of what the bug is.

**Configuration (please complete the following information):**
 - Wheelbase: OSW
 - Rim: Formula
 - Teensy Board: LC
 - Firmware type: USB
 - Firmware version, if known: v1.2 beta.1 and v1.1

**Additional context**
Keep disconnecting issue occurs.
LEDs and 7-Seg. displays are stuttering.

please refer to the videos below.

https://youtu.be/SHrc0STXTC8

https://youtu.be/yca_6G5BIXQ

BTW, Thank you very much for this superb firmware!

darknao, 2018-07-02:
Hi jamesdio.
I can't see the second video, therefore, I'll need a few more information :

jamesdio, 2018-07-02:
Hi darkano.

Thanks for the reply.

Is this happening as soon as you plug the wheel ?
= YES

Beside the leds & screen, does the buttons work correctly ?
= I havn't tested if buttons work or not. I'll check them ASA I get home and let you know. 

Does this issue occur on both stable and beta firmware ?
= 
v1.2beta_1 : stuttering
v1.1 : stuttering

v1.0(attached on firmware folder@GitHub) : working properly

and I tried some other firmware from simracingmachines(https://www.simracingmachines.com/WebShop/software-and-firmware-downloads)

16 JUNE 2018 : stuttering
8 JUNE 2018 : stuttering
4 MAY 2018 : works okay
1 FEB 2018 : works okay

Are you able to revert back to a working firmware ?
=YES

Does this stuttering stop after a few seconds/minutes, or stay like this indefinitely ?
=It loops keeping disconnected and connected continously.

Anyway, I've fixed the link above.You can watch now. so sorry for the inconvenience.

If you need further information, Please ask me.

Thank you.

darknao, 2018-07-03:
Can you try this [firmware ](https://phunk.me/CPAVF4?dl=1) and tell me if it's any better ?

jamesdio, 2018-07-04:
I've tried the new firmware.

Here are the results.

* BMW rim
Buttons work okay.
LEDs work okay.

* Formula rim
_**Buttons work with stuttering**_
LEDs work okay.

https://youtu.be/WU_zroOQO0o

Thank you very much for following up my issue.

jamesdio, 2018-07-06:
I'd looked over for a long time without touching anything at all.

Just staring it. ;)
 
The issue occurs only with my Formula rim.
and there is another strange symptom
such as 7-Segs and LEDs are suddenly displaying unrecognizable words and vibrating for a short momoent without my operation.

This weird phenomenon occurs periodically.

Thanks for your help.

jamesdio, 2018-07-08:
Of course, I can test and bring you results.

When it's ready, Please let me know.

darknao, 2018-07-13:
see [here ](https://github.com/darknao/btClubSportWheel/wiki/Trace-capture)to get the debug firmware and how to extract traces from it.  
You can post the capture file in this issue.

jamesdio, 2018-07-14:
@darknao Here are my logs.

[CSW_FORMULA.txt](https://github.com/darknao/btClubSportWheel/files/2194684/CSW_FORMULA.txt)
[F1.txt](https://github.com/darknao/btClubSportWheel/files/2194685/F1.txt)
[F1_2.txt](https://github.com/darknao/btClubSportWheel/files/2194686/F1_2.txt)
[F1_3.txt](https://github.com/darknao/btClubSportWheel/files/2194687/F1_3.txt)

jamesdio, 2018-07-14:
Here is my log file for BMW which works properly.
I have no idea but this file can be useful for you to debug.

[BMW.txt](https://github.com/darknao/btClubSportWheel/files/2194708/BMW.txt)

darknao, 2018-07-14:
Nice! Thanks!
About F1 files, is that the same wheel? The _2 & _3 looks similar but the first 2 behave very differently

jamesdio, 2018-07-14:
Yes, beacuse I have one formula rim. those logs from one rim.
I'd extracted logs several times to provide you fine samples.

If you meed further information, Please let me know.
Thank you very much for yout hrlp!

darknao, 2018-07-14:
The F1.txt show some serious communication error, like nothing came from your wheel. 
You may have some faulty wiring, or, your wheel start to act strangely.  

Anyway, please try [this firmware](https://phunk.me/YAKGLD?dl=1) and tell me if its any better. Post a new capture file if its still misbehaving.

jamesdio, 2018-07-15:
Thank you for the quick reply!

Here is my new log for latest debugging firmware, rev.1.

[F1_v1.txt](https://github.com/darknao/btClubSportWheel/files/2195407/F1_v1.txt)

darknao, 2018-07-15:
Try this [one](https://phunk.me/KH33XV?dl=1)  
I'm afraid there is not much I can do to solve your issue as your multiple capture files prove this is hardware related.

I highly recommend you to double check your wiring for bad connection, or redo all the soldering and start fresh.  
If there is no improvement after that, I guess you'll have to stick with the old firmware. New firmware only add new wheels support anyway, so you'll not miss anything on your formula.

jamesdio, 2018-07-15:
Thank you for all the help.

I see, I'll solder all of the points from the basis.
and then I'll test it again,

What makes me confuse is the old version works fine but new ones don't.
Don't get me wrong I didn't mean to blame you and your great firmware.
That was just my complaint to myself :(

I'll also check whether there is cold solder connection on my formula rim.

By the way, Here are my new logs.

[F1_v2.txt](https://github.com/darknao/btClubSportWheel/files/2195840/F1_v2.txt)
[F1_v2_1.txt](https://github.com/darknao/btClubSportWheel/files/2195841/F1_v2_1.txt)

darknao, 2018-07-15:
The old firmware is less picky to malformed communication as it supposed to work with only one wheel type. So it doesn't care about how bad the communication is as long as it can read buttons states.  
With the new CSL wheels, the protocol used is a little bit different, and I now need to rely on proper communication to know what kind of wheel is plugged so I can talk is own language.

A software bug is still possible, but since I wasn't able to reproduce it with my wheels, I have to be sure your hardware is not the real issue here.

Last [firmware](https://phunk.me/Y2TL1K?dl=1).  
This is the last modification I can think of that can maybe help :crossed_fingers:

jamesdio, 2018-07-16:
Here is my log for the last one.
I saw the crc error still exists.

[F1_v3.txt](https://github.com/darknao/btClubSportWheel/files/2197963/F1_v3.txt)

I think I have to look over the entire circuit including both teensy and fanatec formula rim.
Thank you for your big help.

I'll let you know as soon as I'll finish the inspection of the circuit connection.

darknao, 2018-07-16:
Did you check if you still have the initial stuttering with this last firmware ?

jamesdio, 2018-07-16:
OMG! I thought the debug revisions were only for debugging.
Because when I flashed the debug firmware FANALEDS doesn't recognize teensy like below.
But It shows on device manager.
All of the debug revisions like the same.

[image]
[image]

darknao, 2018-07-16:
So buttons are still not working correctly with the last firmware...  
I'm afraid I've no more idea on how to fix your issue.

The only suggestion I have is to stick with the old working firmware for now. As I said, there is nothing new for your wheel in this release, so you don't really miss anything.

There is a last thing you can do : make another capture, but play with the buttons while it's running.

jamesdio, 2018-07-17:
Yes, you are right! I think I'd better stick to the old firmware for now.
Eventually, I'm planning to purchase a CSL Mclaren GT3 rim not right now but near future :)

Anyway I played with the buttons but the logs look the same as before.
Here are my logs for both BMW and Formula rims I have.

[BMW DEBUG3 playing with buttons.txt](https://github.com/darknao/btClubSportWheel/files/2202012/BMW.DEBUG3.playing.with.buttons.txt)
[F1 debug3 playing with buttons.txt](https://github.com/darknao/btClubSportWheel/files/2202013/F1.debug3.playing.with.buttons.txt)

BTW I have an idea for my issue.
If it's possible, Can you add an exception code for Formula rim.
The exception code is that It skips CRC error as long as it's formula rim.

That idea was from my imagination from lack of knowledge.
So, you can judge me whether it's bullshit or not ;)

Thank you very much for every single feedback from you!

darknao, 2018-07-17:
Someone else is having the same issue as yours with his BMW wheel, so your not alone!
We're currently investigating, but find out that using another USB cable solve his issue.  
Don't know if this will solve yours, but that worth the try.

jamesdio, 2018-07-18:
I haven't tested the cable trick yet. 

Thank you for the new firmware.
I'm going to test both of them after work. :)

jamesdio, 2018-07-19:
Wow! @darknao your new beta firmware works like a charm!

Both of my rims, BMW and Formula work great.

I haven't tested the cable trick yet because I have no extra usb cable :(
I'll test it later.

Thanks for your BIG BIG help.

Crypto69, 2019-10-03:
I had a similar issue in my recent conversion. I thought it was my wiring so i redid all the wiring from scratch again and had the same result. I finally read this  issues log .

I tried the Teensy LC firmware and it didn’t work. I then tried the beta one above from 19 July 2018

When I installed this beta firmware it started to work. I have no idea what firmware this wheel had as I bought it secondhand. 

Also I’m not sure if its normal but when I plug in the wheel, it seems to boot with the Fanatec Logo ... its takes 30 seconds to a minute after plugging in before any buttons will work in the Game control testing app in the device manager. Is this normal ? 

Fanaleds works and the LCD displays the gears and flashes at gear change.... 

The wheel feels amazing on my Simucube 2 Pro. So far holding up ok. I used a 3D printed adaptor to mount my 70mm Xero Play quick release.  Thanks @darknao