darknao/btClubSportWheel #11: HID Set for BT

ronnywa, 2019-04-09:
Hello,
can you please extend the HID set command for the McLaren GT3?
I have only 37 inputs at the moment.
Have the WT32i and it works very well.

Thank you
Ronny

ronnywa, 2019-04-13:
Hi,

this is the HID Key for 88 Button McLaren GT3 Whell.

HID SET 8f 05010904A1010509190129581500250175019558810205010930093109330934150026FF0075089504810205010939150025073500463B016514750495018142952175048101050A090115002501A1020508094B750195019102C00902A1020508094B750195019102C00903A1020508094B750195019102C00904A1020508094B750195019102C0950D75049101C0 

I have made changes in the program. Now everything is possible via Bluetooth.
Ronny

bonoboxita, 2020-06-15:
Hi Ronnywa,
did you manage to get this working with Bluetooth? can you help me?