darknao/btClubSportWheel #13: Wheel Emulator pinout

Safandula86, 2019-05-31:
Hi, i don't see any information about wheel emulator pinout, can You explain how i must connect teensy 3,2 to Fanatec QR ?

lshachar, 2019-06-02:
Connect miso, mosi, cs (cable select / slave select), clk, gnd. All these connections have same names on both sides, (i.e, you don't need to connect one side's mosi to other side's miso) 
You might want to feed the teensy 3.3V from the 3.3V line on the steering wheel connector, to power it up.

Safandula86, 2019-06-02:
Thank you for help, please correct me if I'm wrong:
Fanatec QR             Teensy 3.2
1.MISO-------------------12
2.MOSI-------------------22
8.CS----------------------10
6.CLK/SCK----------------13

lshachar, 2019-06-03:
I believe you need to switch MISO and MOSI:
1.MISO-------------------11
2.MOSI-------------------12
8.CS----------------------10
6.CLK/SCK----------------13

`
  CORE_PIN13_CONFIG = PORT_PCR_MUX(2);
  CORE_PIN11_CONFIG = PORT_PCR_DSE | PORT_PCR_MUX(2);
  CORE_PIN12_CONFIG = PORT_PCR_MUX(2);
  CORE_PIN10_CONFIG = PORT_PCR_MUX(2);
`

Safandula86, 2019-06-03:
Thank You very much.

closed 2019-06-03 by Safandula86