darknao/btClubSportWheel #14: Where to buy Fanatec Pins

Alevale, 2020-03-22:
**Describe the bug**
This is not a bug per se, but more a place in which I am looking for people that could help me find solutions

**Configuration (please complete the following information):**
 - Rim: BMW GT2

**Additional context**
So as I state, I am looking for pin replacements for my Fanatec wheel because they broke, and I was looking if anyone managed to buy them or reproduce them somehow, without having to pay the 100 euros to fanatec for one of the extra quick release modules.

Any help in this regard is welcome.

Alevale, 2020-03-27:
Just leaving the info here in case this helps anyone, I managed to buy the part from https://www.simracingmachines.com/webshop/fanatec-qr-electrical-pass-through it's not the same (this would be the base connector) but after messaging them they told me they have the 8pins piece that I am looking for as well, just not listed in the website.
Just in case someone runs into this issue like me.
Also, the person that atended me was really nice!

Macieq666, 2020-03-27:
I am looking for pin layout for this cable.
Original was flimsy soldered patched up with elastic glue and seems like it is the reason behind the wheel not working at all

Anyone who has a pin layout i will be thankful

Macieq666, 2020-03-27:
I just contacted them Fanatec and they told me they cant provide diagram layout for pins -lol

So i am more surprise they supply pins with body only as that needs soldering

Btw it is 13 pin but only 8 are occupied

Best Regards

lshachar, 2020-03-27:
> I am looking for pin layout for this cable.
> Original was flimsy soldered patched up with elastic glue and seems like it is the reason behind the wheel not working at all
> 
> Anyone who has a pin layout i will be thankful

I don't understand why you keep posting your question everywhere except where you are really suppose to. simply ### start a new issue### . either in this project or mine. I'm ready to help you out but I won't post my answer where it's not supposed to go.

Macieq666, 2020-03-28:
SHUT UP

EVERYWHERE WHICH MEANS TWO PLACES - I COULD HAVE POSTED IN 20 - WHAT WOULD U SAY THEN?

U CAN KEEP YOUR FUCKING ANSWER TO YOURSELF - I WILL GET IT ANYWAY U CLOWN

Macieq666, 2020-03-28:
AND THE TOPIC IS ABOUT BUYING PINS AND QUESTION ABOUT PINS LAYOUT

VERY FUCKING FAR AWAY FROM EACH OTHER MR SUPER RIGHT

I SEE U HAVE A BIG BUTT PAIN WHENEVER SOMETHING IS NOT THE WAY U WANT IT

U R PATHETIC TO BEGIN

lshachar, 2020-03-28:
wow XD
good luck!

Macieq666, 2020-03-28:
Isacher U R A SHIT HOLE

INSTEAD OF HELPING U R PARADING LIKE A PEACOCK

KEEP YOUR FUCKING PIN LAYOUT - DONT GIVE IT TO ANYONE LIKE FANATEC U ASSHOLE FUCKWIT

I KNOW I WILL HAVE IT VERY SOON U CUNT

Macieq666, 2020-03-28:
AND JUST IN CASE U ASSHOLE I WAS LOOKING FOR PIN LAYOUT NOT YOUR FUCKING STUPID IDEAS

IF THE QUESTION IS IN WRONG PLACE DONT ANSWER

INSTEAD U R TELLING EVERYONE HOW WELL U OBEY RULES U MADE

darknao, 2020-03-28:
Ok, that's enough

closed 2020-03-28 by darknao