darknao/btClubSportWheel #20: teensy 4 wont program

crookeddeebz, 2023-01-06:
anyone have any clue how to program to a 4.0 board?

i tried teensyloader and also the teensyduino forked project both no go.

thanks

crookeddeebz, 2023-01-06:
this is the error i get while  trying to compile as per the dchote release, but on a 4.0 board.
unknown type name 'usb_packet_t'

the darknao release wont work at all with tinyloader,  program doesnt really do anything. auto mode says "add a 2 sec delay, bootloader happened too fast"

thoughts?

crookeddeebz, 2023-01-06:
here is the full error

Arduino: 1.8.19 (Windows 10), TD: 1.57, Board: "Teensy 4.0, Fanatec CSW, 24 MHz, Smallest Code, US English"

fanatec_joystick.c: In function 'fanatec_joystick_send':

fanatec_joystick.c:74: error: unknown type name 'usb_packet_t'

   usb_packet_t *tx_packet;

   ^

fanatec_joystick.c:79: error: 'usb_configuration' undeclared (first use in this function)

     if (!usb_configuration) {

          ^

C:\Users\daveb\Downloads\arduino-1.8.19-windows\arduino-1.8.19\examples\Teensy\fanatecWheelUSB\fanatec_joystick.c:79:10: note: each undeclared identifier is reported only once for each function it appears in

fanatec_joystick.c:84: warning: implicit declaration of function 'usb_tx_packet_count' 

     if (usb_tx_packet_count(JOYSTICK_ENDPOINT) < TX_PACKET_LIMIT) {

         ^

fanatec_joystick.c:84: error: 'JOYSTICK_ENDPOINT' undeclared (first use in this function)

     if (usb_tx_packet_count(JOYSTICK_ENDPOINT) < TX_PACKET_LIMIT) {

                             ^

fanatec_joystick.c:85: warning: implicit declaration of function 'usb_malloc' 

       tx_packet = usb_malloc();

                   ^

fanatec_joystick.c:85: warning: assignment makes pointer from integer without a cast 

       tx_packet = usb_malloc();

                 ^

fanatec_joystick.c:99: error: request for member 'buf' in something not a structure or union

   memcpy(tx_packet->buf, fanatec_joystick_data, JOYSTICK_SIZE);

                   ^

fanatec_joystick.c:99: error: 'JOYSTICK_SIZE' undeclared (first use in this function)

   memcpy(tx_packet->buf, fanatec_joystick_data, JOYSTICK_SIZE);

                                                 ^

fanatec_joystick.c:100: error: request for member 'len' in something not a structure or union

   tx_packet->len = JOYSTICK_SIZE;

            ^

fanatec_joystick.c:102: warning: implicit declaration of function 'usb_tx' 

   usb_tx(JOYSTICK_ENDPOINT, tx_packet);

   ^

fanatec_joystick.c: In function 'fanatec_lights_recv':

fanatec_joystick.c:109: error: unknown type name 'usb_packet_t'

   usb_packet_t *rx_packet;

   ^

fanatec_joystick.c:113: error: 'usb_configuration' undeclared (first use in this function)

     if (!usb_configuration) {

          ^

fanatec_joystick.c:117: warning: implicit declaration of function 'usb_rx' 

     rx_packet = usb_rx(LIGHTS_ENDPOINT);

                 ^

fanatec_joystick.c:117: error: 'LIGHTS_ENDPOINT' undeclared (first use in this function)

     rx_packet = usb_rx(LIGHTS_ENDPOINT);

                        ^

fanatec_joystick.c:125: error: request for member 'buf' in something not a structure or union

   memcpy(buffer, rx_packet->buf, LIGHTS_SIZE);

                           ^

fanatec_joystick.c:125: error: 'LIGHTS_SIZE' undeclared (first use in this function)

   memcpy(buffer, rx_packet->buf, LIGHTS_SIZE);

                                  ^

fanatec_joystick.c:126: warning: implicit declaration of function 'usb_free' 

   usb_free(rx_packet);

   ^

fanatec_joystick.c: In function 'fanatec_lights_available':

fanatec_joystick.c:135: error: 'usb_configuration' undeclared (first use in this function)

   if (!usb_configuration) {

        ^

fanatec_joystick.c:139: warning: implicit declaration of function 'usb_rx_byte_count' 

   count = usb_rx_byte_count(LIGHTS_ENDPOINT);

           ^

fanatec_joystick.c:139: error: 'LIGHTS_ENDPOINT' undeclared (first use in this function)

   count = usb_rx_byte_count(LIGHTS_ENDPOINT);

                             ^

fanatec_joystick.c: In function 'fanatec_lights_recv':

fanatec_joystick.c:129: warning: control reaches end of non-void function 

 }

 ^

unknown type name 'usb_packet_t'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

menekoks, 2023-06-25:
hello - lak